dimalo/klipper-web-control-docker #2: please add version tags to docker images

rpodgorny, 2021-03-06:


dimalo, 2021-03-06:
Not sure which to take for klipper and moonraker as they update themselves from within the container. GitHub actions currently build weekly, should I set the date like 20210306?

montyx99, 2021-03-06:
A version and date together could be fine. The version should shown that when the code in the repo itself modified. The date could give information about when the auto-update modified the repo.

dimalo, 2021-03-06:
> The date could give information about when the auto-update modified the repo.

That is not possible as auto updates happen during runtime of the container. While building the image I can set the build date.
During build the latest master of klipper and moonraker are pulled. The versions then also show in Fluidd.
I think build date as tag makes most sense.

rpodgorny, 2021-03-06:
the main reason for versioned installs is repeatability (and easy reverts) for greater stability. therefore automatic upgrades inside container is definately a no-go. one has to be able to stay on proven stable version forever. this becomes even more important for large deployments.

since both klipper and moonraker do provide versioned releases, i'd expect this version number to be the version corresponding docker image. same goes for mainsail.

if that is not possible, date-as-tag is still slightly better than hash-as-tag solution.

dimalo, 2021-03-06:
Thank you for charing your experience! I agree and I might have not stated it correctly. moonraker offers a plugin which can update klipper's and moonraker's repository. This works well even in the container as long as no dependency updates have to happen as well. Also I cannot prevent users from doing this. But it is not automatic and I did not implement any automation to do updates from within the container. Then I would not even need hash-tags 😄 
Ultimatively I would to have moonraker to be part of klipper in the near future. Then I'm the first one to submit a PR to klipper for this container and only then it is possible to tag with version.
I the meantime I'll go for date. Same for mainsail. happy for anyone to set up a PR if needed...

makuproyect, 2021-03-09:
Alguien ya ha configurado config_another_printer/? (2 3d printer)? editor el archivo moonraker.conf pero no inicia la web del contenedor.

makuproyect, 2021-03-09:
> Alguien ya ha configurado config_another_printer/? (2 3d printer)? editor el archivo moonraker.conf pero no inicia la web del contenedor.

Resuelto port de moonraker debe coincidir con el port en docker-compose
#3dprinter01
[server]
host: 0.0.0.0
port: 7124

#3dprinter02
[server]
host: 0.0.0.0
port: 7125

#3dprinter01
ports:
      - 7124:7124

#3dprinter02
ports:
      - 7125:7125

closed 2021-04-04 by dimalo

dimalo, 2021-04-04:
@rpodgorny I've implemented datetime tags, thank you for the issue!

@makuproyect I'm not an expert on spanish(?) but it seems you want to run multiple printers.
Please have a look at the docker-compose.yml, you just have to add another service with the same docker image, call it differently and bind to the respective /dev, then run docker-compose up -d
You don't need to change anything in moonraker conf. You then have two klipper-moonraker combinations running...

commit 074e9f4c (2021-04-04): fix #2 WIP
commit 64a4e22f (2021-04-04): fix #2

change image tags to datetime string