dimalo/klipper-web-control-docker #3: Moonraker permission issue

montyx99, 2021-03-06:
Hello!
I have the following issue:
```
klipper     | [file_manager.py:register_directory()] - 
klipper     | Moonraker does not have permission to access path (/home/klippy/.config) for (config).
klipper     | [moonraker.py:load_plugin()] - Unable to load plugin (file_manager)
klipper     | Traceback (most recent call last):
klipper     |   File "/home/klippy/moonraker/moonraker/moonraker.py", line 166, in load_plugin
klipper     |     plugin = load_func(config)
klipper     |   File "/home/klippy/moonraker/moonraker/plugins/file_manager.py", line 739, in load_plugin
klipper     |     return FileManager(config)
klipper     |   File "/home/klippy/moonraker/moonraker/plugins/file_manager.py", line 64, in __init__
klipper     |     raise config.error(
klipper     | confighelper.ConfigError: Option 'config_path' is not a valid directory
klipper     | [moonraker.py:main()] - Moonraker Error
klipper     | Traceback (most recent call last):
klipper     |   File "/home/klippy/moonraker/moonraker/moonraker.py", line 166, in load_plugin
klipper     |     plugin = load_func(config)
klipper     |   File "/home/klippy/moonraker/moonraker/plugins/file_manager.py", line 739, in load_plugin
klipper     |     return FileManager(config)
klipper     |   File "/home/klippy/moonraker/moonraker/plugins/file_manager.py", line 64, in __init__
klipper     |     raise config.error(
klipper     | confighelper.ConfigError: Option 'config_path' is not a valid directory
klipper     | 
klipper     | During handling of the above exception, another exception occurred:
klipper     | 
klipper     | Traceback (most recent call last):
klipper     |   File "/home/klippy/moonraker/moonraker/moonraker.py", line 663, in main
klipper     |     server = Server(system_args, file_logger)
klipper     |   File "/home/klippy/moonraker/moonraker/moonraker.py", line 108, in __init__
klipper     |     self._load_plugins(config)
klipper     |   File "/home/klippy/moonraker/moonraker/moonraker.py", line 137, in _load_plugins
klipper     |     self.load_plugin(config, plugin)
klipper     |   File "/home/klippy/moonraker/moonraker/moonraker.py", line 172, in load_plugin
klipper     |     raise ServerError(msg)
klipper     | utils.ServerError: Unable to load plugin (file_manager)
klipper     | [moonraker.py:main()] - Server Shutdown
klipper     | 2021-03-06 11:03:35,299 INFO exited: moonraker (exit status 1; not expected)
klipper     | 2021-03-06 11:03:36,303 INFO gave up: moonraker entered FATAL state, too many start retries too quickly
klipper     | 2021-03-06 11:03:36,301 - supervisord_dependent_startup - [INFO   ] 
klipper     | 2021-03-06 11:03:36,302 - supervisord_dependent_startup - [INFO   ] New event: Service moonraker went from STARTING to BACKOFF
```
What can cause this issue?

dimalo, 2021-03-06:
> Moonraker does not have permission to access path (/home/klippy/.config) for (config).

How did you set up the config folder?
seems like the klippy user cannot access the path.

If it is mounted from the Pi, please make sure you have permissions set so that the user within the container can write to it.

On the Pi in the config folder
`ls -lsa`

If it is only owned by your Pi user, I guess the klippy user inside does not have access.

you can issue chmod 666 to all files in the folder, be careful not to set this for secrets!
`chmod 666 <config_folder_path>/*`

montyx99, 2021-03-06:
I've logged into the container and if I performed a `chown klippy:klippy` on both of the `.config` and `gcode_files` without recursion made that the moonraker was able to start. This is not the best solution, because this modified the permissions outside of the docker too.
I'm thinking about a better solution. If it's come up, I'm going to let you know.

montyx99, 2021-03-06:
What if the volumes are temp volumes during the install? Attach the volumes, copy the files to another folder with the correct permission, then detach the volumes.
The printer.cfg can be uploaded through a web frontend.

dimalo, 2021-03-06:
I guess having the folder created upfront in the Pi is better. Also to run the containers not as sudo or root but add the Pi user to the docker group. I just checked my install, most files are 644 and still possible to write them. owner is the Pi user but docker-compose up -d was also run by this user. perhaps that does the trick...

montyx99, 2021-03-06:
ohh, so you elevated the PI user to run docker commands, am I right? it's in the docker group. I see. Hmmm I need to figure it out alone in that case :)

montyx99, 2021-03-06:
Currently I use chmod 666. I'm going to think about further solution later. I'm going to close the issue. Thanks

closed 2021-03-06 by montyx99

dimalo, 2021-03-06:
You can also chown 1001:1001 on the config folder in the Pi and add the Pi user to group 1001, then at least apply 664

montyx99, 2021-03-06:
I've done that before, but thanks :+1: