dimalo/klipper-web-control-docker #4: Would it be possible to control rpi gpio for PSU control?

BugsonPL, 2021-03-25:
I tried many ways but failed.

dimalo, 2021-04-12:
@BugsonPL please try with the latest main branch. I've included the gpiolib [here](https://github.com/dimalo/klipper-web-control-docker/commit/ae4aeb5ad29a2ff7b75f885e9ca4c339ccacc66b#diff-6127beb2ead01fe933482621ea2f78cc8fd31bf3f9bf499cda67deac600b414eR54)
Look for the gpio device in /dev
```bash
# something like /dev/gpiochip0 perhaps?
ls -lsa /dev/*
```
then add a device mapping to docker-compose.yml

You have to map to :/dev/gpiochip0 as [moonraker checks for that one](https://github.com/Arksine/moonraker/blob/b111e3d6c73e38b1ec3231a7299b8735b703daae/moonraker/components/power.py#L254)
```yaml
  klipper:
    image: dimalo/klipper-moonraker
    build:
      dockerfile: ./klipper/Dockerfile
      context: .
      # args:
      #   DEVICE_GROUP: device
      #   DEVICE_GID: 987
    container_name: klipper
    ports:
      - 7125:7125
    restart: unless-stopped
    volumes:
      - gcode_files:/home/klippy/gcode_files
      # be aware to create your own branch if you mount the config folder as it will be updated on the main branch
      # that way you can merge upstream changes to your customized configs
      - ./config:/home/klippy/.config
      # - <<your_config_path>>:/home/klippy/.config
      # - ./printer.cfg:/home/klippy/.config/printer.cfg
    # mount serial device - take care to grant sufficient permissions to the device: <host_dev>:<container_dev>
    # put <container_dev> into your printer.cfg
    devices:
      - /dev/ttyUSB0:/dev/ttyUSB0
      - /dev/gpiochip0:/dev/gpiochip0 # <---- HERE
```

Also check permissions on the gpio device and make sure they are the same as for the printers serial device.
Let me know if it worked :-)

dimalo, 2021-04-23:
Closing this, please let me know if the issue persists...

closed 2021-04-23 by dimalo