dimalo/klipper-web-control-docker #7: Support for arm32v6 Raspberry Pi?

ahalekelly, 2021-04-25:
Raspberry Pi is by far the most popular way to run Klipper, but only the Raspberry Pi 4 supports arm32v7. Do you what it would take to compile a docker image for the arm32v6 architecture to support the Pi 3 and Pi Zero W?

ahalekelly, 2021-04-25:
Well I added linux/arm/v6 to the github workflow file in my fork, the docker images built without errors and are up on https://hub.docker.com/u/ahalekelly, but I'm still getting error 139 when I try to run them.

ahalekelly, 2021-04-26:
Well I couldn't figure out how to make sure I was running the right image, because dockers's local sha256 hashes don't match the ones on dockerhub. So I deleted all images other than the ARMv6 version from my dockerhub, and now it works? How do you make sure docker downloads the right version? For both the ARMv6 and ARMv7 images, `docker image inspect` simply says `"Architecture": "arm"`

ahalekelly, 2021-04-26:
Looks like I'm not the first to encounter this:
https://github.com/diyhue/diyHue/issues/444
https://github.com/docker/for-linux/issues/1025

ahalekelly, 2021-04-26:
Well I figured out how to check the image versions against Docker Hub. 
`docker manifest inspect --verbose ahalekelly/mainsail:20210426-1045`
returns all the images for that tag, each with the structure
```
        {
                "Ref": "docker.io/ahalekelly/mainsail:latest@sha256:07d135fc505c1b40b31eaa2278f2397c6a810a08d2226603fda33262d071ffb8",
                "Descriptor": {
                        "mediaType": "application/vnd.docker.distribution.manifest.v2+json",
                        "digest": "sha256:07d135fc505c1b40b31eaa2278f2397c6a810a08d2226603fda33262d071ffb8",
                        "size": 2400,
                        "platform": {
                                "architecture": "arm",
                                "os": "linux",
                                "variant": "v6"
                        }
                },
                "SchemaV2Manifest": {
                        "mediaType": "application/vnd.docker.distribution.manifest.v2+json",
                        "schemaVersion": 2,
                        "config": {
                                "mediaType": "application/vnd.docker.container.image.v1+json",
                                "digest": "sha256:77561174a806bf645cc9d446e471caf7f644767ce0964e1bcd833e5d4dfbfc0d",
                                "size": 8927
                        },
                    ...
```
The first digest matches the digest from dockerhub (07d135fc505c), and the second digest matches the Image ID from `docker image ls` (77561174a806)

It's pulling the correct version now, not sure what I did that fixed it...

dimalo, 2021-04-27:
Hey @ahalekelly, really good investigation!
When you had the wrong arch image did you later try to pull the correct arch but same tag? Normally docker does not pull a tag if it is in local cache already (even having the wrong arch I guess).
So having built the correct one, while issuing _docker pull_ maybe docker just did not pull at all in the first place. Now having new tags to pull the correct arch in the first place could explain why it just works...
Big thanks for trying out! Would be interesting how much the little raspi is able to pull off.
Are you planning on having multiple printers connected to it?

ahalekelly, 2021-04-27:
Thanks! Good thinking, but I don't think that's it, I deleted all the docker images several times in my troubleshooting. And there was never a build of my repo that didn't include the v6 arch.

The Pi Zero apparently doesn't have enough power for Octoprint, but lots of people run Klipper + Mainsail/Fluidd on it. I'm setting up a print farm, one Pi Zero W per printer, and a central frontend server. I'll see if I can replicate a solution to this issue on another Pi Zero.

closed 2021-04-27 by dimalo

dimalo, 2021-04-27:
added the arch to the builds, thanks again for trying it out!
Yeah I've seen like tenfold less resources used running klipper and fluidd compared to octoprint.

Just one last question out of curiosity, how do you plan to update all of those raspis?

ahalekelly, 2021-04-28:
Ok, new Raspberry Pi install, I was able to confirm that Docker 18.09 still pulls the ARMv7 image, and Docker 20.10 pulls the ARMv6 image. Fluidd doesn't have an ARMv6 image so it pulls the ARMv7 image and fails at runtime, but Klipper and Mainsail run.

Haven't thought much about updates yet. Docker should make automatic updates much easier though.

commit 2dbcb2d9 (2021-04-27): fix #7

add ARMv6 to multiarch build