dimalo/klipper-web-control-docker #8: Cannot SAVE_CONFIG

montyx99, 2021-04-27:
I had permission issues before with the .config folder. The Moonraker was not able to reach the config folder. I've used the given advice with luck, but I've just figured it out, I cannot save the config online.
Such as I performed a bed mesh leveling and used the SAVE_CONFIG command, and the printer.cfg hasn't been modified. I think that there is an issue with the Klipper container, just I cannot figure it out.

dimalo, 2021-04-27:
Unfortunately I don't have a better solution for you as in #3 
Please check ownership and permissions on folder and files & run as non-root

montyx99, 2021-05-06:
> Unfortunately I don't have a better solution for you as in #3
> Please check ownership and permissions on folder and files & run as non-root

Can you help me, what is the user, that the Moonraker uses? I thought that it is the same like the Klipper itself, but it has already permission in case of config folder.

dimalo, 2021-05-15:
It is the same user (klippy)
No issues for me whatsoever so closing this issue as it seems to be a config issue on your side...

closed 2021-05-15 by dimalo