dimalo/klipper-web-control-docker #10: Fluidd UI not working propperly

darknblack, 2021-06-22:
Hi

I can access the printer via moonraker but not via Fluidd

[image]

I'm getting: Authorization error

and this: 
`{"error": {"code": 404, "message": "Not Found", "traceback": "Traceback (most recent call last):\n\n  File \"/home/klippy/moonraker-env/lib/python3.9/site-packages/tornado/web.py\", line 1681, in _execute\n    result = self.prepare()\n\n  File \"/home/klippy/moonraker/moonraker/app.py\", line 786, in prepare\n    raise tornado.web.HTTPError(404)\n\ntornado.web.HTTPError: HTTP 404: Not Found\n"}}`

whenever accessing the http://{moonraker_ip}/access

And this too in Fluidd
[image]

dimalo, 2021-06-26:
hi @darknblack,
maybe this is because my mainsail docker image has an included reverse proxy to moonrakers endpoints. So you can set up the printer with mainsails URL.
Using Fluidd's official image this functionality is not included though the klipper image has to expose it's port and has to be properly configured. In fluidd you enter moonraker's URL. This did not work well for me as well with fluidd.
That's why I switched back to build my own.

Fow nor you can as well, have a look into docker-compose, there is a commented out build section for fluidd.
this builds a similar image as mainsail, but with fluidd. fixed setup issues for me, as I only have to enter fluidd's URL as printer URL.

For more than one printer, this is a bit more tricky though.

dimalo, 2021-07-13:
@darknblack not sure if your issue is fixed.

I did go to own fluidd builds as I like the reverse proxy config which saves a lot of time in setting up CORS etc...

closed 2021-07-13 by dimalo