dimalo/klipper-web-control-docker #11: Moonraker : Permission Denied (database)

AktasC, 2021-07-12:
Hello there 👋 

First of all, thanks for you work  
I've been trying to install & run this stack but here are the different errors I kept getting :   

```bash
klipper    | [moonraker.py:load_component()] - Unable to load component: (database)
klipper    | Traceback (most recent call last):
klipper    |   File "/home/klippy/moonraker/moonraker/moonraker.py", line 200, in load_component
klipper    |     component = load_func(config)
klipper    |   File "/home/klippy/moonraker/moonraker/components/database.py", line 518, in load_component
klipper    |     return MoonrakerDatabase(config)
klipper    |   File "/home/klippy/moonraker/moonraker/components/database.py", line 74, in __init__
klipper    |     os.mkdir(self.database_path)
klipper    | PermissionError: [Errno 13] Permission denied: '/home/klippy/.moonraker/database'
klipper    | [moonraker.py:main()] - Moonraker Error
klipper    | Traceback (most recent call last):
klipper    |   File "/home/klippy/moonraker/moonraker/moonraker.py", line 200, in load_component
klipper    |     component = load_func(config)
klipper    |   File "/home/klippy/moonraker/moonraker/components/database.py", line 518, in load_component
klipper    |     return MoonrakerDatabase(config)
klipper    |   File "/home/klippy/moonraker/moonraker/components/database.py", line 74, in __init__
klipper    |     os.mkdir(self.database_path)
klipper    | PermissionError: [Errno 13] Permission denied: '/home/klippy/.moonraker/database'
klipper    | 
klipper    | During handling of the above exception, another exception occurred:
klipper    | 
klipper    | Traceback (most recent call last):
klipper    |   File "/home/klippy/moonraker/moonraker/moonraker.py", line 755, in main
klipper    |     server = Server(app_args, file_logger)
klipper    |   File "/home/klippy/moonraker/moonraker/moonraker.py", line 138, in __init__
klipper    |     self._load_components(config)
klipper    |   File "/home/klippy/moonraker/moonraker/moonraker.py", line 176, in _load_components
klipper    |     self.load_component(config, component)
klipper    |   File "/home/klippy/moonraker/moonraker/moonraker.py", line 206, in load_component
klipper    |     raise ServerError(msg)
klipper    | utils.ServerError: Unable to load component: (database)
klipper    | [moonraker.py:main()] - Server Shutdown
klipper    | 2021-07-11 23:42:21,495 INFO exited: moonraker (exit status 1; not expected)
```  

I've seen the #3 issue and decided to try and run a shell to `chmod 664` as advised.    

Now the issue seems to be an EACCESS but somewhere else.   
The message I keep getting over and over :   

```bash
klipper    | 2021-07-11 23:48:08,240 INFO spawned: 'klipper' with pid 10
klipper    | supervisor: couldn\'t exec /home/klippy/klippy-env/bin/python: EACCES
klipper    | supervisor: child process was not spawned
klipper    | 2021-07-11 23:48:08,246 INFO exited: klipper (exit status 127; not expected)
```

---------------

Setup : 
- Docker
- Docker-Compose
- Portainer
- Ubuntu 20.04

AktasC, 2021-07-13:
After investigating a bit more I've found an underlying issue : 
`/home/klippy/.moonraker` is owned by root, a simple `chown klippy:klippy /home/klippy/.moonraker` fixed it.  

Now Mainsail works but Fluidd still can't communicate with Moonraker.

dimalo, 2021-07-13:
Thanks for your thorough investigation!
yep sorry for that, I might have forgotten to change the supplied moonraker config. There the correct db path should be .moonraker_database
I changed this on behalf of moonraker example config stating it should be kept separate to other moonraker data but forgot to update the config.
the path prepared in the Docker image matches the example shown [here](https://moonraker.readthedocs.io/en/latest/configuration/)

dimalo, 2021-07-13:
Those two COPY ops are by intent as root. It's not always a bad thing :)

Please have a look into the logs with docker-compose logs -f or similar. Is there anything about the mcu?
Seems like you really did a great job on fixing your issues, but now it's more of a communication issue.
The only problem with docker containers and communication I experienced so far was mismatched group permissions.
Fixed it by introducing argument based config in docker-compose.yml
      # args:
      #   DEVICE_GROUP: device
      #   DEVICE_GID: 987

But I'm pretty sure you checked already right?

AktasC, 2021-07-15:
> Those two COPY ops are by intent as root. It's not always a bad thing :)
> 
> Please have a look into the logs with docker-compose logs -f or similar. Is there anything about the mcu?
> Seems like you really did a great job on fixing your issues, but now it's more of a communication issue.
> The only problem with docker containers and communication I experienced so far was mismatched group permissions.
> Fixed it by introducing argument based config in docker-compose.yml
> # args:
> # DEVICE_GROUP: device
> # DEVICE_GID: 987
> 
> But I'm pretty sure you checked already right?
> 
> I would be interested if you can make sure if the mcu is actually running...

Thanks for your answer and potential leads  

I tried with the args for `DEVICE_GROUP` & `DEVICE_GID`, unfortunately it didn't change anything.  

I've scouted the logs but the only error message I get besides `systemctl` not being available is this one :  
``` 
[moonraker.py:_check_ready()] - ,
Printer is not ready,
The klippy host software is attempting to connect.  Please,
retry in a few moments.
```

Also, `sudo systemctl list-units | grep active` returns a loaded and active `ttyUSB0-tty-ttyUSB0.device`

dimalo, 2021-07-25:
https://discord.gg/8UcjUuf6

ahalekelly, 2021-08-05:
I just pulled the latest version of the git repo and am also getting this error now. Did you guys ever figure it out? That discord invite is expired.

dimalo, 2021-08-06:
@ahalekelly check the moonraker config. I just pushed a change to align .moonraker_database with the touched file in the Dockerfile. Sorry for the inconvenience, wanted to stick to what moonraker docs suggest...

closed 2021-08-06 by dimalo

ahalekelly, 2021-08-07:
Great, thanks! It's working for me again. Looks like you forgot to make this update in docker-compose.yml though?