dimalo/klipper-web-control-docker #15: klipper-moonraker container doesn't work

watzon, 2021-09-29:
Seems as thought he klipper-moonraker container is broken. I tested `latest` as well as the one that was pushed yesterday, but I'm getting the python error:

``
Traceback (most recent call last):
  File "/home/klippy/klipper/klippy/klippy.py", line 8, in <module>
    import util, reactor, queuelogger, msgproto
  File "/home/klippy/klipper/klippy/reactor.py", line 6, in <module>
    import os, gc, select, math, time, logging, Queue as queue
ModuleNotFoundError: No module named 'Queue'
```

This is a common error when trying to run a python2 project with python3.

dimalo, 2021-09-30:
hi @watzon - yes I noticed as well. Seems like omitting the PYTHONHOME env variable from moonraker.ini does the trick, whilst for klipper and python2 it has to be set...
Can you try the latest :next tag image?
Just pulling onto a raspi3b...

watzon, 2021-09-30:
That seems to have done the trick!

closed 2021-09-30 by watzon