dimalo/klipper-web-control-docker #21: klipper version tags

pciavald, 2021-11-14:
Hi,

Would it be possible to include klipper's version (and possibly moonraker) in the klipper-moonraker image's tags corresponding to the klipper repo's tags for the builds that are not just daily builds of the master branch?

That would be convenient to be able to follow the main releases.

for instance:
```
klipper-moonraker:0.10.0-0.7.1
```

Thank you!

dimalo, 2021-11-15:
great idea! will look into it, if you have any suggestions, I'd like to hear them :)

pciavald, 2021-11-15:
To avoid doing it manually, it seems there's a way to trigger github actions from another repo if you get collaboration from the remotes : https://github.community/t/triggering-by-other-repository/16163 and you could then get the tag name with something like this : https://github.community/t/how-to-get-just-the-tag-name/16241

pciavald, 2021-11-15:
If that can be of interest, currently i'm using a modified version of your docker images that clones like:
```dockerfile
ENV VERSION v0.10.0
RUN git clone \
  --depth 1 \
  --branch ${VERSION} \
  https://github.com/Klipper3d/klipper
```

dimalo, 2022-01-18:
sorted
https://github.com/dimalo/klipper-web-control-docker/blob/main/klipper/Dockerfile#L4

closed 2022-01-18 by dimalo