dimalo/klipper-web-control-docker #26: Configuration File Not Found: '/home/klippy/.config/moonraker.conf'

NamkingWIN, 2022-01-11:
I am stuck at this point, the moonraker start and shutdown. Please see below

```
RESULT 2
klipper     | OK2022-01-11 18:42:19,363 - supervisord_dependent_startup - [INFO   ] No more processes to start for initial startup, ignoring all future events.
klipper     | 2022-01-11 18:42:19,576 INFO exited: dependentstartup (exit status 0; expected)
klipper     | 2022-01-11 18:42:19,576 INFO exited: moonraker (exit status 1; not expected)
klipper     | 2022-01-11 18:42:20,584 INFO spawned: 'moonraker' with pid 18
klipper     | [utils.py:setup_logging()] - config_file: /home/klippy/.config/moonraker.conf
klipper     | [utils.py:setup_logging()] - log_file: /var/log/klipper/moonraker.log
klipper     | [utils.py:setup_logging()] - software_version: v0.7.1-285-g505c1bf
klipper     | [moonraker.py:main()] - Moonraker Error
klipper     | Traceback (most recent call last):
klipper     |   File "/home/klippy/moonraker/moonraker/moonraker.py", line 867, in main
klipper     |     server = Server(app_args, file_logger, event_loop)
klipper     |   File "/home/klippy/moonraker/moonraker/moonraker.py", line 77, in __init__
klipper     |     self.config = config = confighelper.get_configuration(self, args)
klipper     |   File "/home/klippy/moonraker/moonraker/confighelper.py", line 409, in get_configuration
klipper     |     raise ConfigError(
klipper     | confighelper.ConfigError: Configuration File Not Found: '/home/klippy/.config/moonraker.conf''
klipper     | [moonraker.py:main()] - Server Shutdown
klipper     | 2022-01-11 18:42:21,780 INFO success: moonraker entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
klipper     | 2022-01-11 18:42:21,781 INFO exited: moonraker (exit status 1; not expected)
klipper     | 2022-01-11 18:42:22,789 INFO spawned: 'moonraker' with pid 21
klipper     | [utils.py:setup_logging()] - config_file: /home/klippy/.config/moonraker.conf
klipper     | [utils.py:setup_logging()] - log_file: /var/log/klipper/moonraker.log
klipper     | [utils.py:setup_logging()] - software_version: v0.7.1-285-g505c1bf
klipper     | [moonraker.py:main()] - Moonraker Error
klipper     | Traceback (most recent call last):
klipper     |   File "/home/klippy/moonraker/moonraker/moonraker.py", line 867, in main
klipper     |     server = Server(app_args, file_logger, event_loop)
klipper     |   File "/home/klippy/moonraker/moonraker/moonraker.py", line 77, in __init__
klipper     |     self.config = config = confighelper.get_configuration(self, args)
klipper     |   File "/home/klippy/moonraker/moonraker/confighelper.py", line 409, in get_configuration
klipper     |     raise ConfigError(
```

DrFatalis, 2022-01-12:
Check that the config folder is mounted correctly in the docker-compose.yml and as a test, chmod -R 777 /config
If volume is mounted, it could be the read/write rights that are not okay.

NamkingWIN, 2022-01-12:
Yea, I fixed it with chmod config folder

closed 2022-01-12 by NamkingWIN