dimalo/klipper-web-control-docker #28: Mainsail Settings Storage

cron410, 2022-01-14:
Where are the settings stored for Mainsail? When I `docker-compose down` and then `docker-compose up -d` again the webcam settings and GUI preferences are wiped out. I tried finding any possible modified files in the container with `find / -type d -mmin -2 -print` directly after adding my webcams, but it doesn't seem like the settings are stored anywhere in the container filesystem.

dimalo, 2022-01-18:
Mainsail settings are stored in your browsers local storage (think of it like cookies)
so if you clean your browser's cache they're gone or if you visit mainsail only through private browsing windows.
try setting the url as favorite. I'm mainly using fluidd but settings stay same, at least on the same browser.

closed 2022-01-18 by dimalo

cron410, 2022-01-18:
Yea I finally got Fluidd working by manually adding a printer, it wasn't
defaulted to port 7125.

Opera doesn't clear any cookies that I know of, but i think I'm going to
give up on Mainsail and keep using Fluidd like I was before moving to
Docker.