dimalo/klipper-web-control-docker #30: DBus Connection Not available, systemd service tracking and actions are disabled

Robnex, 2022-01-31:
Moonraker Warnings:
[machine]: DBus Connection Not available, systemd service tracking and actions are disabled

Hello,
i installed your docker container today and when i move to the mainsail dashboard i have the Moonraker warnings on top.

Are there any fixes?

Greetings

SlLVERKlNG, 2022-03-08:
Were you able to figure this out? I'm having the same issue.

SlLVERKlNG, 2022-03-09:
There was a discussion about another klipper docker project.
https://github.com/mkuf/prind/issues/8

His modification to the moonraker.conf resolved my issue by adding

[machine]
provider: none

erustyrannus, 2022-11-02:
Can confirm that removed the warning for me.

```
[machine]
provider: none
```

added to moonraker.conf removed the warning.

PR #39 (2022-12-15): Develop
- fix #35 
- fix #37
- fix #30
closed 2022-12-15 by dimalo