dimalo/klipper-web-control-docker #31: Portainer edition

Robnex, 2022-01-31:
Hello,

would it be possible to create a docker_compose_yml for portainer. With the actual one it dont open port 7125 for klipper.

Greetings

DonKracho, 2022-03-12:
First of all, thank you for the great work.

I have the same issue, but I'm not sure if we need a special compose for prtainer. Im using potainer 2.11.1 on a MeCool M8S Pro W with Armbian installed. I copied the docker-compose.yml content into a custom template and deployed the stack. The fluidd and mainsail containers are running but try to connect to an invalid printer <IP of the device>:8010 respective 8011. Mainsail is unusabe. For Fluidd you can try to add a new printer. An extemal printer is working, but the local klipper container does not answer. 

I copied the config files to the moonraker_data volume using WinSCP, but this does not help either. There is no answer to http://<local ip>:7125/server/info 

I can not get into a container console, the container seems to be restarting continually. The log lists permission errors:
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/supervisor/loggers.py", line 98, in emit
    msg = self.fmt % record.asdict()
  File "/usr/local/lib/python3.10/site-packages/supervisor/loggers.py", line 287, in asdict
    now = time.time()
PermissionError: [Errno 1] Operation not permitted
Error: Cannot open an HTTP server: socket.error reported errno.EPERM (1)
For help, use /usr/local/bin/supervisord -h

My docker skills are not sufficient. Any hints what to do?

DonKracho, 2022-03-14:
I added a  'privileged: true'  line to the klipper service. Not the smartest idea but klipper is able to start now and I can access the container console. I had to copy all config files to .config but these files are not visible in fluidd.

Is the compose.yml missing to propagate the moonraker_data volume to the fluidd and mainsail services?
 
Finally moonraker can not access the .config folder. I did a  'chmod -r a+w .config' but that did not help. Still get a warning:
 
Moonraker Warnungen gefunden.
Moonraker does not have permission to access path (/home/klippy/.config) for (config).
[machine]: DBus Connection Not available, systemd service tracking and actions are disabled

piter-pit, 2022-10-31:
I'm also fighting with this docker to work.
similar case:
1. Klipper starts
2. Moonraker fail - missing config file, after coping starts
3. copied printer.cfg not visible neither in fluidd nor mainsail, in both unable to load the file

erustyrannus, 2022-11-02:
Here is my Docker-compose, the container you need to worry about is the Klipper-Moonraker one, the others do not have any data (itself). If you clone this repo you can check the Config folder (I have just redirected it to another path and all is working)

`version: '3.4'
services:
  klipper:
    image: dimalo/klipper-moonraker:latest
    container_name: klipper
    environment:
      - TZ=Europe/Madrid #Change to your Timezone
    ports:
      - 7125:7125
    volumes: #Change path accordingly
      - (Path you want)/gcode_files:/home/klippy/gcode_files
      - (Path you want)/klipper/config:/home/klippy/.config
      - (Path you want)/Docker/klipper/moonraker_data:/home/klippy/.moonraker
    devices:
      - /dev/serial/by-id/device:/dev/serial/by-id/device #You can find this with this command:  ls /dev/serial/by-id/*
    restart: unless-stopped
    
  fluidd:
    image: cadriel/fluidd:latest
    container_name: fluidd
    environment:
      - TZ=Europe/Madrid #Change to your Timezone
    ports:
      - 8010:80
    depends_on: 
      - klipper
    links:
      - klipper:klipper
      
  mainsail:
    image: dimalo/mainsail:latest
    container_name: mainsail
    environment:
      - TZ=Europe/Madrid #Change to your Timezone
    ports:
      - 8011:80
    depends_on: 
      - klipper
    links:
      - klipper:klipper
    restart: unless-stopped
`

I am still trying to get the webcam to work properly, I have managed to print without issues.

I also recommend you to have a deep read here:
https://moonraker.readthedocs.io/en/latest/configuration/ since this is what controls everything aparently.

PS: I am also new to Klipper.

erustyrannus, 2022-11-02:
> Hello,
> 
> would it be possible to create a docker_compose_yml for portainer. With the actual one it dont open port 7125 for klipper.
> 
> Greetings

Here is my Docker-compose, the container you need to worry about is the Klipper-Moonraker one, the others do not have any data (itself). If you clone this repo you can check the Config folder (I have just redirected it to another path and all is working) 

Be sure to copy all the "config" files from this repo to your new path together with your printer.cfg and adjust them following the documentation (cited below).

```
version: '3.4'
services:
  klipper:
    image: dimalo/klipper-moonraker:latest
    container_name: klipper
    environment:
      - TZ=Europe/Madrid #Change to your Timezone
    ports:
      - 7125:7125 #You can replace the host port by changing the first value (EG:  XXXX:7125) to any free port on the host.
    volumes: #Change path accordingly
      - (Path you want)/klipper/gcode_files:/home/klippy/gcode_files
      - (Path you want)/klipper/config:/home/klippy/.config
      - (Path you want)/klipper/moonraker_data:/home/klippy/.moonraker
    devices:
      - /dev/serial/by-id/device:/dev/serial/by-id/device #You can find this with this command:  ls /dev/serial/by-id/*
    restart: unless-stopped
    
  fluidd:
    image: cadriel/fluidd:latest
    container_name: fluidd
    environment:
      - TZ=Europe/Madrid #Change to your Timezone
    ports:
      - 8010:80
    depends_on: 
      - klipper
    links:
      - klipper:klipper
      
  mainsail:
    image: dimalo/mainsail:latest
    container_name: mainsail
    environment:
      - TZ=Europe/Madrid #Change to your Timezone
    ports:
      - 8011:80
    depends_on: 
      - klipper
    links:
      - klipper:klipper
    restart: unless-stopped
```
Ensure that the host folder permissions are correct (allowing the user to write, read and execute)

I am still trying to get the webcam to work properly, I have managed to print without issues.

I also recommend you to have a deep read here:
https://moonraker.readthedocs.io/en/latest/configuration/ since this is what controls everything aparently.

PS: I am also new to Klipper.

piter-pit, 2022-11-02:
Hi,
The Docker-compose file is basically the same, but mine is not working.
Maybe it's related to https://github.com/dimalo/klipper-web-control-docker/issues/41 - have to check. Currently I've installed outside the docker as a regular install and it works fine.

erustyrannus, 2022-11-02:
> Hi, The Docker-compose file is basically the same, but mine is not working. Maybe it's related to #41 - have to check. Currently I've installed outside the docker as a regular install and it works fine.

For your error I would check the folder permissions on the host. looks like the issue is that.

piter-pit, 2022-11-07:
> > Hi, The Docker-compose file is basically the same, but mine is not working. Maybe it's related to #41 - have to check. Currently I've installed outside the docker as a regular install and it works fine.
> 
> For your error I would check the folder permissions on the host. looks like the issue is that.

I've chmoded files and folders and even set privileged mode so it must be something different.
I think I will stay with installed directly at least for now, maybe in near future I will go with second attempt ;)