dimalo/klipper-web-control-docker #32: Is z_calibration plugin supported?

NamkingWIN, 2022-02-01:
Hi,
I have done with the guide at https://github.com/protoloft/klipper_z_calibration#preconditions
But the moonraker log show below:
`[update_manager client z_calibration]
path = /home/klippy/klipper_z_calibration
origin = https://github.com/protoloft/klipper_z_calibration.git
2022-02-01 12:10:59,968 [shell_command.py:_check_proc_success()] - Command (git -C /home/klippy/klipper_z_calibration status -u no) successfully finished
2022-02-01 12:10:59,978 [shell_command.py:_check_proc_success()] - Command (git -C /home/klippy/klipper_z_calibration config --get branch.master.remote) successfully finished
2022-02-01 12:10:59,988 [shell_command.py:_check_proc_success()] - Command (git -C /home/klippy/klipper_z_calibration remote get-url origin) successfully finished
2022-02-01 12:11:02,523 [git_deploy.py:initialize()] - Git Repo z_calibration: Initialization failure`

How do I fix that?

dimalo, 2022-09-25:
I'm sure it has to do with permissions on the folder. make sure the klippy user has write permissions

dimalo, 2022-09-25:
@NamkingWIN please check with imagetag :next

I did not test as I don't have a suitable printer - please leave a feedback if the extension and updates work for you

commit 177c43b1 (2022-09-25): ADD klipper_z_calibration extension (resolve #32)
closed 2022-12-15 by dimalo