dimalo/klipper-web-control-docker #34: Can't get multiple klipper/moonrakers to connect to fluidd

napter, 2022-03-27:
I've been trying for days to get two klippers running. I used the sample docker-compose.yml but still run into issues. Here is the latest version of my docker-compose.yml.

```
version: '3.4'

services:
  klipper:
    image: dimalo/klipper-moonraker
    build:
      dockerfile: ./klipper/Dockerfile
      context: .
      # args:
      #   DEVICE_GROUP: device
      #   DEVICE_GID: 987
    container_name: klipper
    ports:
      - 7125:7125
    restart: unless-stopped
    volumes:
      - gcode_files:/home/klippy/gcode_files
      # be aware to create your own branch if you mount the config folder as it will be updated on the main branch
      # that way you can merge upstream changes to your customized configs
      - ./config:/home/klippy/.config
      - moonraker_data:/home/klippy/.moonraker
      # - <<your_config_path>>:/home/klippy/.config
      # - ./printer.cfg:/home/klippy/.config/printer.cfg
    # mount serial device - take care to grant sufficient permissions to the device: <host_dev>:<container_dev>
    # put <container_dev> into your printer.cfg
    devices:
      - /home/nathan/docker/usbmap/usb5.4.2:/dev/ttymxc3

  ##################################################################
  # Example config for a second printer on a different serial port
  ##################################################################

  klipper_top:
    image: dimalo/klipper-moonraker
    build:
      dockerfile: ./klipper/Dockerfile
      context: .
      # args:
      #   DEVICE_GROUP: device
      #   DEVICE_GID: 987
    container_name: klipper_top
    ports:
      # As 7125 is already used by the first printer, we map this moonraker's port to 7126 on the host
      - 7126:7125
    restart: unless-stopped
    volumes:
      # Let both klipper instances use the same storage for gcode files
      - gcode_files:/home/klippy/gcode_files
      # be aware to create your own branch if you mount the config folder as it will be updated on the main branch
      # that way you can merge upstream changes to your (developed) configs...
      - ./config_top:/home/klippy/.config
      #- ./config_top/moonraker_data:/home/klippy/.moonraker #NATHAN: not sure if this can be shared or not
      - moonraker_data:/home/klippy/.moonraker
      # - <<your_config_path>>:/home/klippy/.config
      # - ./another_printer.cfg:/home/klippy/.config/printer.cfg
    # mount serial device - take care to grant sufficient permissions to the device: <host_dev>:<container_dev>
    # put <container_dev> into your printer.cfg
    devices:
      - /home/nathan/docker/usbmap/usb5.4.3:/dev/ttymxc3

  fluidd:
    #image: dimalo/fluidd
    #image: cadriel/fluidd
    build:
        dockerfile: Dockerfile
        context: frontend
    restart: unless-stopped
    container_name: fluidd
    ports:
      - 81:80
    depends_on: 
      - klipper
      - klipper_top
    links:
      - klipper:klipper
      - klipper_top:klipper_top

  mainsail:
    image: dimalo/mainsail
    build:
      dockerfile: Dockerfile
      context: frontend
      args:
        FRONTEND_ZIP_URL: https://github.com/meteyou/mainsail/releases/latest/download/mainsail.zip
    container_name: mainsail
    ports:
      - 8011:80
    depends_on: 
      - klipper
      - klipper_top
    links:
      - klipper:klipper
      - klipper_top:klipper_top

volumes: 
  gcode_files:
  moonraker_data:

```

Fluidd somehow is connecting to the 7125 moonraker although I am not sure how it knows to do that. Fluidd is hosted on ubuntu port 81. When I load up fluidd under printers it shows it's own address under printers - shouldn't it show the moonraker web server at 7126? 
[image]

I then try to add the second moonraker at 7126 (http://ubuntu:7126/) but it gives me an error "blocked by CORS policy". However, I am able to access the same URL from my browser without an error. The moonraker.conf in both ./config folders are the same. Interestingly if I try and add the first printer at 7125 it also gives me the cors error. However, it is clearly connected to that printer because I am can controll it.

[image]

I am sure I am missing something fundamental here but I cannot figure out what it is. Any help would be greatly appreciated.

TheSeven, 2022-03-28:
This is a moonraker config issue.
Try setting this in your `moonraker.conf` file:
```[authorization]
cors_domains:
  *
```
That will allow any web site to interact with that klipper instance, not just one hosted on the same host/port.
For security reasons it may be better to put a list of specific allowed host names there, however those did not appear to be matched quite the way that I expected, so I can't tell you which specific strings you would need to put there.

dimalo, 2022-06-24:
FInally I got the time to reply... :(
as @TheSeven stated, moonraker can be configured to allow any URL. you can specify "ubuntu" as alllowed CORS source.
have a look at the config documentation [here](https://moonraker.readthedocs.io/en/latest/configuration/#authorization)
- add localhost and 127.0.0.1 as trusted clients
- add ubuntu as CORS origin like "http://ubuntu:*" so you get both ports (or specify both)
Please leave a feedback if it worked out! Cheers

dimalo, 2022-06-24:
oh and the answer why fluidd magically works: I've configured nginx in the fluidd/mainsail images to reverse proxy to the first klipper instance - see [here](https://github.com/dimalo/klipper-web-control-docker/blob/main/frontend/upstreams.conf.template) and [here](https://github.com/dimalo/klipper-web-control-docker/blob/main/frontend/client.conf)