dimalo/klipper-web-control-docker #37: Can't perform input shaper tuning due to bad klippy-env

HitLuca, 2022-08-31:
I just finished tuning my klipper input shaper, but not without issues.
Looks like the provided python env in `~/klippy-env` has issues, more specifically it returns errors when running `klippy-env/bin/pip install numpy` (something similar to `no module named packaging.utils` from `pip._vendor.packaging`). This issue made me try everything to get it working, but at the end the simplest solution was to recreate it with

```python
pip install virtualenv
rm -r /home/klippy/klippy-env
virtualenv --python="/usr/bin/python2.7" klippy-env
```

after those commands the numpy installation succeeded!

HitLuca, 2022-08-31:
Just to ease the following steps, make sure to install numpy and matplotlib on the system interpreter too using pip, otherwise you'll need to run the `klipper/scripts/calibrate_shaper.py /tmp/resonances_#_*.csv -o /tmp/shaper_calibrate_#.png` command by specifying the python interpreter in `klippy-env`

dimalo, 2022-09-22:
just added numpy & matplotlib as global pip installs, with klipper now also supporting python3 this s a lot more straightforward

dimalo, 2022-09-22:
just reopened to make sure the release builds, then will close and reference in release notes...

PR #39 (2022-12-15): Develop
- fix #35 
- fix #37
- fix #30
closed 2022-12-15 by dimalo