dimalo/klipper-web-control-docker #41: Problems with upload

basn, 2022-10-21:
Hi,
been trying to get this to work, seems like its not up to date? i got alot of errors when i finaly got it running.
and i keep having issues with uploading that it seems not to be enabled;
HTTPServerRequest(protocol='http', host='fluidd.local:8010', method='POST', uri='/server/files/upload', version='HTTP/1.0', remote_ip='192.168.x.x')
Traceback (most recent call last):
  File "/home/klippy/moonraker-env/lib/python3.10/site-packages/tornado/web.py", line 1681, in _execute
    result = self.prepare()
  File "/home/klippy/moonraker/moonraker/app.py", line 882, in prepare
    fm.check_write_enabled()
  File "/home/klippy/moonraker/moonraker/components/file_manager/file_manager.py", line 199, in check_write_enabled
    raise self.server.error(
utils.ServerError: Write access is currently disabled.  Check notifications for warnings.

Did i mess something up when trying to fix stuff? What should i look for?

serjs, 2022-10-31:
For latest releases of Moonraker you need ot disable [validate_services ](https://moonraker.readthedocs.io/en/latest/configuration/#machine)dut ValidationError if you run Moonraker in docker container.

Try validate_service: False in moonraker.conf [machine] block, also check other warnings in Moonraker, that can prevent write access using new Config validation process.

rpodgorny, 2022-12-05:
i'm also experiencing this. modification of moonraker.conf as advised did not help. what should i try next?