dimalo/klipper-web-control-docker #43: WebCam with timelapse configuration

erustyrannus, 2022-11-01:
Hello, can you tell me if this project is implemented in your build?

https://github.com/mainsail-crew/moonraker-timelapse

truong0vanchien, 2022-11-04:
You need to add timelapse manually via docker container.
Otherwise, you can add some lines in Dockerfile of Klipper as below:

```
# Install ffmpeg.
RUN apt-get install -y --no-install-recommends ffmpeg

# Install wget.
RUN apt-get install -y wget

# Put timelapse.py to docker container. (timelapse.py is from moonraker timelapse repo)
COPY timelapse.py ${HOME}/moonraker/moonraker/components/timelapse.py
```

p0pp3t0n, 2023-02-22:
@erustyrannus Did you make it work? My cameras are not being detected by mainsail. I know they are working since i took images using ffmpeg from the container.

erustyrannus, 2023-02-23:
> @erustyrannus Did you make it work? My cameras are not being detected by mainsail. I know they are working since i took images using ffmpeg from the container.

Hi, no.  I decided to install on a SBC using [Kiauh](https://github.com/th33xitus/kiauh)

closed 2023-02-23 by erustyrannus