dimalo/klipper-web-control-docker #45: Permission denied: '/home/klippy/.config/moonraker.conf' leads to 502 Bad Gateway issue

moookino, 2022-11-16:
Hi,

I've tried to pull latest version and started getting 502 Bad Gateway on fluidd and mainsail, after investigation it seems they're not able to connect moonraker

From container startup log:
`klipper   | 2022-11-16 11:42:03,142 INFO spawned: 'moonraker' with pid 70
klipper   | [utils.py:setup_logging()] - config_file: /home/klippy/.config/moonraker.conf
klipper   | 2022-11-16 11:42:03,730 - supervisord_dependent_startup - [INFO   ]
klipper   | 2022-11-16 11:42:03,730 - supervisord_dependent_startup - [INFO   ] New event: Service moonraker went from BACKOFF to STARTING
klipper   | [utils.py:setup_logging()] - startup_warnings: []
klipper   | [utils.py:setup_logging()] - log_file: /var/log/klipper/moonraker.log
klipper   | [utils.py:setup_logging()] - software_version: v0.7.1-660-g5d856b9
klipper   | [utils.py:setup_logging()] - python_version: 3.10.7 (main, Sep 13 2022, 03:02:33) [GCC 10.2.1 20210110]
klipper   | [moonraker.py:main()] - Moonraker Error
klipper   | Traceback (most recent call last):
klipper   |   File "/home/klippy/moonraker/moonraker/confighelper.py", line 924, in _parse_file
klipper   |     stat = file_path.stat()
klipper   |   File "/usr/local/lib/python3.10/pathlib.py", line 1097, in stat
klipper   |     return self._accessor.stat(self, follow_symlinks=follow_symlinks)
klipper   | PermissionError: [Errno 13] Permission denied: '/home/klippy/.config/moonraker.conf'
klipper   |
klipper   | During handling of the above exception, another exception occurred:
klipper   |
klipper   | Traceback (most recent call last):
klipper   |   File "/home/klippy/moonraker/moonraker/moonraker.py", line 460, in main
klipper   |     server = Server(app_args, file_logger, event_loop)
klipper   |   File "/home/klippy/moonraker/moonraker/moonraker.py", line 67, in __init__
klipper   |     self.config = config = self._parse_config()
klipper   |   File "/home/klippy/moonraker/moonraker/moonraker.py", line 129, in _parse_config
klipper   |     config = confighelper.get_configuration(self, self.app_args)
klipper   |   File "/home/klippy/moonraker/moonraker/confighelper.py", line 1040, in get_configuration
klipper   |     source.read_file(start_path)
klipper   |   File "/home/klippy/moonraker/moonraker/confighelper.py", line 1028, in read_file
klipper   |     self._parse_file(main_conf, [])
klipper   |   File "/home/klippy/moonraker/moonraker/confighelper.py", line 1012, in _parse_file
klipper   |     if not file_path.is_file():
klipper   |   File "/usr/local/lib/python3.10/pathlib.py", line 1322, in is_file
klipper   |     return S_ISREG(self.stat().st_mode)
klipper   |   File "/usr/local/lib/python3.10/pathlib.py", line 1097, in stat
klipper   |     return self._accessor.stat(self, follow_symlinks=follow_symlinks)
klipper   | PermissionError: [Errno 13] Permission denied: '/home/klippy/.config/moonraker.conf'
klipper   | [moonraker.py:main()] - Server Shutdown
klipper   | 2022-11-16 11:42:03,742 - supervisord_dependent_startup - [INFO   ] Services:
klipper   | 2022-11-16 11:42:03,751 - supervisord_dependent_startup - [INFO   ]  - klipper           RUNNING                         dependent_startup: True
klipper   | 2022-11-16 11:42:03,761 - supervisord_dependent_startup - [INFO   ]  - moonraker         STARTING                        dependent_startup: True   wait_for: 'klipper:RUNNING'
klipper   | 2022-11-16 11:42:03,768 - supervisord_dependent_startup - [INFO   ] Services not yet running (1): moonraker
`
Have something changed in the past docker updates?

speendo, 2022-11-18:
I run mainsail but I seem to be affected by the same issue.
It seems like something in Moonraker has changed and now it's not working anymore...

moookino, 2022-11-22:
I've ended up using some older version from docker build as a workaround, but having issues with config as well, so put all needed items in the printer.cfg

moookino, 2022-11-24:
@speendo I ended up migrating to https://github.com/mkuf/prind, seems like more robust and better maintained solution with same set of  feature/frontends

speendo, 2022-11-24:
Thanks. I was using prind before, it sort of worked but this project was easier to handle for me.
Well, at least until recently :)

@dimalo: do you have any plans about this issue? (just to make it clear: I am fine if you don't have any, just want to know)

speendo, 2022-12-03:
I think I might have understood where the problem is coming from.

Recent versions of moonraker expect all files to be in the directory `/home/<user>/printer_data/`, e.g. `/home/<user>/config` or `/home/<user>/gcodes`.

This docker image however is configured to store all files at `/home/klippy` (where klippy is the user name defined in the Dockerfile). So the config lays in `/home/klippy/.config` and gcodes files are at `/home/klippy/gcode_files`.

For an ideal solution, it would be best to follow the standard, that moonraker expects. However, to do this, one would not only need to change the Dockerfile (which would be okay), but also many people who linked their gcode or settings folder in the `docker-compose.yml` would need to change their settings.

For the moment, I decided for a quick workaround (that I actually would need to repeat every time I update my docker image, so take care). It works like this

1. Go to `klipper/moonraker.ini` and change the line `command=/home/klippy/moonraker-env/bin/python /home/klippy/moonraker/moonraker/moonraker.py -l /var/log/klipper/moonraker.log -c /home/klippy/.config/moonraker.conf` to `command=/home/klippy/moonraker-env/bin/python /home/klippy/moonraker/moonraker/moonraker.py -l /var/log/klipper/moonraker.log -d /home/klippy -c /home/klippy/.config/moonraker.conf`
2. Go into the running docker container with `docker exec -it klipper /bin/bash`
3. In the docker container delete the directories `/home/klippy/printer_data/config` and `/home/klippy/printer_data/gcodes`. Make sure that you backup the content of those folders, in case they are not empty. You have been warned! To delete a folder, use `rm -r <folder_path>
4. Create symlinks to replace the folders you just deleted: Run `ln -s /home/klippy/.config /home/klippy/printer_data/config` and `ln -s /home/klippy/gcode_files /home/klippy/printer_data/gcodes`.
5. Restart the docker container.
6. In case you still have problems, they might be caused by permission problems. Make sure that the user klippy has read and write access to the folders you just linked (you can check by logging into the docker container again and run `ls -al /home/klippy`).

It would be possible to integrate this workaround in the dockerfile, however it might be better to think of a solution that is consistent with the moonraker default configuration.

speendo, 2022-12-09:
The problems should be solved with my pull request or by using https://github.com/speendo/klipper-web-control-docker/tree/main in the mean time

sammcj, 2023-01-11:
@dimalo any update on getting this fix merged in?