dimalo/klipper-web-control-docker #52: Provide the container for unraid

eRaz3r, 2023-02-12:
Hello,

is there any possibility to provide or adapt the contasiner for use with unraid? I don't have a clue how to use it there. 

Thanks in advance

coltography, 2024-05-30:
+1 for this, much needed

nrgbistro, 2024-07-20:
While waiting for this, I'd recommend looking into installing [Portainer](https://www.portainer.io/) to give your unraid system more flexibility. Alternatively, there is the [Docker Compose Manager](https://forums.unraid.net/topic/114415-plugin-docker-compose-manager/) plugin.

rmcmilli, 2025-01-28:
I was able to get it working on unraid with the exception of the USB camera in fluidd. 

You'll need compose manager as stated by @nrgbistro . The default compose file uses build but I don't think unraid supports that by default so I built the klipper/moonraker container and stored it in a local registry, the fluidd container works fine from pull. 

I already use a local docker registry for various other customized images so I didn't need to set this part up. It might be a bit of effort. In short the klipper-moonraker image needs to be already built and available. 

You'll need to create a host directory for your files for persistence -- I used /mnt/user/appdata/klipper -- and map that location instead of the default in the compose file. 

Confirm the  ports in the compose file don't conflict with any other containers and save. Compose up and you should see a klipper and fluidd container in the list. 

------- my compose  --------

#version: '2'

services:
  klipper:
    # image: dimalo/klipper-moonraker
    image: local-registry/klipper-moonraker
    # build:
    #   dockerfile: ./klipper/Dockerfile
    #   context: .
    # if klipper cannot connect to the printer check permissions on rpi then add the group here
    # group_add:
    #   - "996"
    #   - "998"
    cap_add:
      - SYS_NICE
    container_name: klipper
    ports:
      - 7125:7125
      #- 8011:8011
    restart: unless-stopped
    volumes:
      # - gcode_files:/home/klippy/gcode_files
      - gcode_files:/home/klippy/printer_data/gcodes
      # be aware to create your own branch if you mount the config folder as it will be updated on the main branch
      # that way you can merge upstream changes to your customized configs
      # - ./config:/home/klippy/.config
      - /mnt/user/appdata/klipper:/home/klippy/.config
      # - /mnt/user/appdata/klipper:/home/klippy/printer_data/config
      # - /mnt/user/appdata/klipper:/home/klippy/config
      - moonraker_data:/home/klippy/.moonraker_database
      # - <<your_config_path>>:/home/klippy/.config
      # - ./printer.cfg:/home/klippy/.config/printer.cfg
    # mount serial device - take care to grant sufficient permissions to the device: <host_dev>:<container_dev>
    # put <container_dev> into your printer.cfg
    devices:
      - /dev/ttyUSB0:/dev/ttyUSB0
      # - /dev/video0
      # - /dev/video1
    privileged: true
    cpu_shares: 2048
#networks:
 # klipper-bridge:
 #   driver: bridge
 #   internal: true

  ##################################################################
  # Example config for a second printer on a different serial port
  ##################################################################

  # klipper_another_printer:
  #   image: dimalo/klipper-moonraker
  #   build:
  #     dockerfile: ./klipper/Dockerfile
  #     context: .
  #     # args:
  #     #   DEVICE_GROUP: device
  #     #   DEVICE_GID: 987
  #   container_name: klipper_another_printer
  #   ports:
  #     # As 7125 is already used by the first printer, we map this moonraker's port to 7126 on the host
  #     - 7126:7125
  #   restart: unless-stopped
  #   volumes:
  #     # Let both klipper instances use the same storage for gcode files
  #     - gcode_files:/home/klippy/gcode_files
  #     # be aware to create your own branch if you mount the config folder as it will be updated on the main branch
  #     # that way you can merge upstream changes to your (developed) configs...
  #     - ./config_another_printer:/home/klippy/.config
  #     # - <<your_config_path>>:/home/klippy/.config
  #     # - ./another_printer.cfg:/home/klippy/.config/printer.cfg
  #   # mount serial device - take care to grant sufficient permissions to the device: <host_dev>:<container_dev>
  #   # put <container_dev> into your printer.cfg
  #   devices:
  #     - /dev/ttyACM0:/dev/ttyACM0

  fluidd:
    image: dimalo/fluidd
    # build:
    #   dockerfile: Dockerfile
    #   context: frontend
    restart: unless-stopped
    container_name: fluidd
    ports:
      - 8010:80
      - 8011:8080
    devices:
      - /dev/video0
      - /dev/video1
    depends_on: 
      - klipper
      # - klipper_another_printer
    links:
      - klipper:klipper
      # - klipper_another_printer:klipper_another_printer

  # mainsail:
  #   image: dimalo/mainsail
  #   build:
  #     dockerfile: Dockerfile
  #     context: frontend
  #     args:
  #       FRONTEND_ZIP_URL: https://github.com/meteyou/mainsail/releases/latest/download/mainsail.zip
  #   container_name: mainsail
  #   ports:
  #     - 8011:80
  #   depends_on: 
  #     - klipper
  #     # - klipper_another_printer
  #   links:
  #     - klipper:klipper
  #     # - klipper_another_printer:klipper_another_printer

volumes: 
  gcode_files:
  moonraker_data: