dimalo/klipper-web-control-docker #59: Unable to connect to mcu

Ed1ks, 2023-05-06:
I am failing to connect to the printer. After following the guide and testing several thing i now hope someone has an idea how to fix this problem.
My system is an Raspberry pi 4 8gb with rocky linux and docker.
Another SD-Card with klipper and mainsailOS installed barebones works perfect, but this setup with Docker cant establish the connection to the printer.
It always returns the error "mcu 'mcu': Unable to connect" and quits.

This is my current docker-compose file (yes it has to many parameters, but these result from testing)
`    klipper:
        image: dimalo/klipper-moonraker
        container_name: klipper
        restart: unless-stopped
        privileged: true
        device_cgroup_rules:
            - 'c 188:* rmw'
        group_add:
            - "188"
            - "0"
        build:
            dockerfile: ./klipper/Dockerfile
            context: .
        cap_add:
            - SYS_NICE
        ports:
            - 7125:7125
        devices:
            - '/dev:/dev'
        volumes:
            - klipper_gcode_files:/home/klippy/gcode_files
            - klipper_config:/home/klippy/.config
            - klipper_moonraker_config:/home/klippy/.moonraker
            - /dev/serial/by-id/:/dev/serial/by-id/
            #- /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0:/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0`

Some information for the USB Port:
`$ ls -lsa /dev/ttyUSB0
0 crw-rw----. 1 root dialout 188, 0 Feb 28 00:00 /dev/ttyUSB0

$ ls /dev/serial/by-id/ 
usb-1a86_USB_Serial-if00-port0
`

I added the container log and the printer.cfg (both as txt)

[klipper_logs.txt](https://github.com/dimalo/klipper-web-control-docker/files/11413131/klipper_logs.txt)
[printer.cfg.txt](https://github.com/dimalo/klipper-web-control-docker/files/11413147/printer.cfg.txt)

drk944, 2023-05-22:
I'm running into similar issues as you, were you ever able to figure it out?
Here's the output of my serial/by-id command
`$ ls /dev/serial/by-id/`
`usb-Klipper_stm32g0b1xx_43004E001750425938323120-if00`

Ed1ks, 2023-05-23:
Sadly I still cant get it to work

eirinnm, 2023-06-03:
@Ed1ks you could try mounting the entire dev folder as a volume.

You currently have `- '/dev:/dev'` in your _devices_ mounting, but AFAIK that isn't supported. Only specific devices can be mounted, not the whole folder, and you can't use wildcards either.

But what you can do is mount the /dev folder as a _volume_. This means your container can see all your host devices, but it can't access them because it doesn't have permissions. I haven't yet figured out how to give permissions to certain devices, but my workaround at the moment is to give permissions to all devices with privileged: true (which you are already doing).

So to recap, this is what I've done:
In docker-compose:
    privileged: true
    volumes:       - /dev:/dev

In klipper config:
[mcu]
serial: /dev/serial/by-id/usb-Klipper_stm32f103xe_37FFD9054246383416870857-if00

eirinnm, 2023-06-04:
Here's another approach. This one doesn't mount the entire devices folder, it just mounts the printer.

in docker-compose:
privileged: true
devices:
     - /dev/serial/by-id/usb-Klipper_stm32f103xe_37FFD9054246383416870857-if00:/dev/ttyUSB0

in klipper config:
[mcu]
serial: /dev/ttyUSB0

This has the advantage of being more secure (your container can't access other host devices) but has the disadvantage of failing on container startup if the printer isn't connected.

eirinnm, 2023-06-10:
A final, most secure and most robust option. Mounts all devices, but only gives access to hardware. Privileged mode isn't needed.

```
services:
  klipper:
    image: dimalo/klipper-moonraker
    privileged: false #not needed now
    device_cgroup_rules:
      - 'c *:* rmw' #this gives the container permission to access hardware
    build:
      dockerfile: ./klipper/Dockerfile
      context: .
    cap_add:
      - SYS_NICE
    container_name: klipper
    ports:
      - 7125:7125
    restart: unless-stopped
    volumes:
      - gcode_files:/home/klippy/printer_data/gcodes
      - ./config:/home/klippy/printer_data/config
      - moonraker_data:/home/klippy/.moonraker_database
      - /dev:/hostdevices # mount host devices inside the container without permissions
```

bencorkeryhome, 2024-04-20:
> Here's another approach. This one doesn't mount the entire devices folder, it just mounts the printer.
>
> in docker-compose: privileged: true devices: - [...]

This is the one that worked for me. Thank you!